YosysHQ/eqy #37: Use with net names generated by ghdl-yosys-plugin

fayalalebrun, 2023-06-18:
I am trying to use eqy in conjuction with [ghdl-yosys-plugin](https://github.com/ghdl/ghdl-yosys-plugin). However, I get the following error:
```
ERROR: conflicting matches for gold bit $auto$ghdl.cc:806:import_module$28: $auto$ghdl.cc:806:import_module$100 vs $auto$ghdl.cc:806:import_module$42
```
There are many such auto-generated names in the netlist, so the next thing I tried was to add the following:
```
[match *]
nodefault
gold-match /^((?!import_module).)*$/
```
However, this has no effect on the error.

I am not sure if preventing those signals from being matched would fix the issue, or what the implications would be.

Therefore, I am wondering whether it is possible to use eqy with ghdl-yosys-plugin?  If not, is this a limitation of eqy, or of ghdl-yosys-plugin?

jix, 2023-07-24:
Can you share the complete `.eqy` file you are using?

fayalalebrun, 2023-07-30:
I was using it for a cache controller which is rather complex. However, I transcribed one of the examples to vhdl and it seems to fail with a similar error:
https://gist.github.com/fayalalebrun/055de56382081b62b47fe2639ae8e000

jix, 2023-09-20:
I reduced it further to the following eqy file directly containing two smaller netlists that reproduce this without needing to use the ghdl plugin: https://gist.github.com/jix/ff72778c03cd6769a277b3c98658441e

**gold**
[image]

**gate**
[image]

The cells named "13" and "2" have the same name although they were unrelated even in the ghdl example.

@clairexen Any idea how that gets the partitioning code to produce conflicting drivers?

clairexen, 2023-09-27:
First attempt at a fix for this issue:
- Use https://github.com/YosysHQ/yosys/tree/claire/bufnorm
- Replace `insbuf -chain` in eqy with `bufnorm -bits`