YosysHQ/eqy #45: Widely differing behavior with different standard cell libs (but the same design)

openroadie, 2023-09-13:
I used the advice provided in #40 to successfully run a simple design with ASAP7 (runtime ~ 1m). But when I run the same design with two other public libraries (nangate45 and skyhd130), I hit #40 for skyhd130 and the following error for nangate45:

QY 10:37:58 [eqy_test] Warning: Cannot find entity _*_.*.
EQY 10:37:59 [eqy_test] partition: starting process "cd eqy_test; yosys -ql partition.log partition.ys"
EQY 10:38:08 [eqy_test] partition: Warning: Wire gate.aes_cipher_top.us31.a.4.\_27162_.A3 is used but has no driver.
EQY 10:38:08 [eqy_test] partition: Warning: Wire gate.aes_cipher_top.us31.a.4.\_26838_.A4 is used but has no driver.
EQY 10:38:08 [eqy_test] partition: Warning: Wire gold.aes_cipher_top.us31.a.4.\_26897_.A3 is used but has no driver.
EQY 10:38:08 [eqy_test] partition: ERROR: Found 3 problems in 'check -assert'.
EQY 10:38:08 [eqy_test] partition: finished (returncode=1)

Test case has 3 directories for the three cases. 

Please let me know if there is anything else I can provide/help with to make the debug easier :).

[bug.tar.gz](https://github.com/YosysHQ/eqy/files/12600538/bug.tar.gz)

openroadie, 2023-09-13:
Please ignore the nangate45 error for now. I had a problem with the library cells. I will upload a new testcase once I have fixed it and done some more debugging.

jix, 2023-09-15:
With the following EQY configuration the skyhd130 check doesn't hit #40 and also passes quickly. I think the main difference is the way the cell libraries are structured, here there were a lot of unused signals after flattening because the cell definitions from the PDK define signals they then don't actually use. I left inline comments explaining why I chose the commands / options I've used:

```
[gold]
read_verilog -sv ./before.v ./formal_pdk.v

[gate]
read_verilog -sv ./after.v ./formal_pdk.v

[script]
prep -top aes_cipher_top -flatten

# Using `rename -hide` is a better performing choice than nomatch if the signal names have no meaning at all
rename -hide */_*_.*

# This removes unused signals before partitioning so no partitions are created for them
opt_clean -purge

memory_map

[collect *]
# This groups signals like `some_signal[0]`, `some_signal[1]`, ... that only differ in the index
group *[] \1[]

[strategy basic]
use sat
depth 2
```

openroadie, 2023-09-15:
@jix Thanks for the script. I believe it is running into #46 for now since adding this line:

`rename -hide */_*_.*`

Gets me: 

```
Traceback (most recent call last):
  File "/usr/local/bin/eqy", line 1208, in <module>
    main()
  File "/usr/local/bin/eqy", line 1180, in main
    make_partitions(ctx.args, ctx, ctx.job)
  File "/usr/local/bin/eqy", line 731, in make_partitions
    partition_ids(cfg)
  File "/usr/local/bin/eqy", line 669, in partition_ids
    for lhs, rhs in search_entities(ctx, ctx.matched_ids[module_match], ctx.matched_ids[module_match],
  File "/usr/local/bin/eqy", line 558, in search_entities
    rhs = Pattern(other_expr, groups, groupdict)
  File "/usr/local/bin/eqy", line 509, in __init__
    p.regex = self.shell_pattern_to_regex(pattern)
  File "/usr/local/bin/eqy", line 478, in shell_pattern_to_regex
    return re.compile("".join(regex))
  File "/usr/lib/python3.8/re.py", line 252, in compile
    return _compile(pattern, flags)
  File "/usr/lib/python3.8/re.py", line 304, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/lib/python3.8/sre_compile.py", line 764, in compile
    p = sre_parse.parse(p, flags)
  File "/usr/lib/python3.8/sre_parse.py", line 948, in parse
    p = _parse_sub(source, state, flags & SRE_FLAG_VERBOSE, 0)
  File "/usr/lib/python3.8/sre_parse.py", line 443, in _parse_sub
    itemsappend(_parse(source, state, verbose, nested + 1,
  File "/usr/lib/python3.8/sre_parse.py", line 834, in _parse
    p = _parse_sub(source, state, sub_verbose, nested + 1)
  File "/usr/lib/python3.8/sre_parse.py", line 443, in _parse_sub
    itemsappend(_parse(source, state, verbose, nested + 1,
  File "/usr/lib/python3.8/sre_parse.py", line 549, in _parse
    raise source.error("unterminated character set",
re.error: unterminated character set at position 8
```

jix, 2023-09-15:
Yeah that is the error fixed by #46. Without `group *[] \1[]` the error also shouldn't appear as it's on the code path for that, but whether you encounter the error also depends on the signal names in the design.

jix, 2023-09-20:
@openroadie with #46 now merged and the approach I outlined above, is there anything remaining for this issue? If you still need to fully test it, just let me know when you've done so, so I can close this issue (or respond to anything else that might come up).

openroadie, 2023-09-20:
@jix Thank you for all your help. There is nothing else remaining for this issue but I am sure I will need more help going forward :). My runtime for this was around 7m which is great.

closed 2023-09-20 by openroadie