YosysHQ/eqy #55: eqy stuck in infinite loop with gate design primitives (Xilinx gates) as blackboxes

xtofalex, 2024-02-17:
Hi,
I guess I'm using naively the tool by comparing two netlists: initial RTL and one file after synthesis targeting Xilinx and providing Xilinx gates as blackboxes.
Still when doing so, the tool hangs in an infinite loop in the eqy execute, in particular in co_flatten_worker method.

I've attached a test case.

Versions I'm using:
Yosys 0.38+54 (git sha1 f8d4d7128, clang 15.0.0 -fPIC -Os)
eqy: latest (5791c90)

[test_case.tar.gz](https://github.com/YosysHQ/eqy/files/14319678/test_case.tar.gz)

Thanks.
Christophe

maliberty, 2024-04-09:
It would be good if there was some response here.  Christophe's work will eventually benefit the OSS tool flow but needs to be able to be verified with eqy.

jix, 2024-04-09:
Comparing blackboxes on one side to logic on the other side is not supported. This should produce an error, though, and not hang, so this is an error handling bug.

closed 2026-10-21 by xtofalex