YosysHQ/eqy #58: Support for UDP?

maliberty, 2024-03-20:
In asap7 the cells description in Verilog come from Liberate which uses UDP like:

```
primitive altos_latch (q, v, clk, d);
	output q;
	reg q;
	input v, clk, d;

	table
		* ? ? : ? : x;
		? 1 0 : ? : 0;
		? 1 1 : ? : 1;
		? x 0 : 0 : -;
		? x 1 : 1 : -;
		? 0 ? : ? : -;
	endtable
endprimitive
```

but I get errors  when reading as soon as it sees this:
```
platforms/asap7/work_around_yosys/asap7sc7p5t_SEQ_RVT_TT_220101.v:37: ERROR: syntax error, unexpected TOK_ID
```

Are UDP supported in eqy?  This is a common characterization tool so I expect this will be a common issue.

maliberty, 2024-03-20:
This arises when trying to use the suggested script in #57.  Without UDP (which were commented out, presumably because they don't parse) I get lots of errors about undriven signals.

jix, 2024-03-20:
The lack of UDP support in EQY is entirely due to the lack of UDP support in yosys and there are currently no plans to add UDP support to yosys.

maliberty, 2024-03-20:
That's unfortunate.  So I would have to manually write Verilog models for all the registers?  Will  `always @(posedge clock)` style work ok in eqy?

jix, 2024-03-20:
Everything that our FV flows support is supported in EQY, which includes almost all of what yosys can synthesize so `always @ (en or d)  if (en) q <= d;` would describe a dlatch and `always @(posedge clk) ...` a dff, etc. following the patterns described in IEEE 1364.1.

For designs that use multiple clocks (which includes posedge and negedge of the same clock signal) or any async logic besides async resets (that includes latches) it may be necessary to use the `clk2fflogic` pass at the end of the gate and gold scripts. Such a design may work without that pass, but in that case would correspond to an implicit assumption that all clocks are the same global clock and that any async logic is an async reset for FFs using that clock. In the context of EQY this would then mean that any differences that are hidden by that assumption would not be detected.

This is described here https://yosyshq.readthedocs.io/projects/ap011/en/latest/faq_sby.html#clock-signals although as that was written for SBY it mentions the `multiclock on` option which corresponds to adding `clk2fflogic` in EQY.

---

Instead of using the verilog simulation models, which usually aren't written using synthesizable verilog and thus require manual edits, it might also be possible to use `read_liberty` if there is a corresponding liberty file that defines the cell behavior and which yosys can read.

I'm not too familiar with the liberty file format, and even less familiar with the organization of the asap7 PDK, but this seems to mostly work with the latches and FFs when running this to convert some of the asap7 liberty files into synthesizable verilog that should work with EQY:

```
yosys -p 'read_liberty -ignore_miss_func asap7sc7p5t_SEQ_RVT_TT_ccs_201020.lib; write_verilog'
```

I had to add `-ignore_miss_func` as `read_liberty` as the liberty files are missing the `function` node on the internal `IQ` pins of the clock gating cells, but fixing this in the liberty files might be easier or at least easier to maintain than manually writing synthesizable verilog cell definitions. If not, converting the cells where this is not an issue from liberty might still reduce the amount of cells that require writing verilog manually.

jix, 2024-03-20:
After using the following awk script to add a function definition to the `IQ` pins the example liberty file I picked loads completely:

```
gawk '{print};/ pin \(IQ\)/ {print "function : \"ENA + SE\";"}' asap7sc7p5t_SEQ_RVT_TT_ccs_201020.lib > fixed.lib
```

maliberty, 2024-03-20:
If we can use Liberty instead of Verilog to describe the cells that would be great!  Can you give an example script for that?

maliberty, 2024-03-20:
nvm I see you mean you can do a one-time conversion, not use it directly in eqy

jix, 2024-03-20:
You can also directly use read_liberty from EQY. The gate and gold scripts in EQY support all yosys commands. The `read_liberty` command creates a yosys RTLIL module for each defined cell (given that function definitions are present in the liberty file), the same way that `read_verilog` would create a yosys RTLIL module for each verilog module. I merely illustrated it using a stand alone yosys invocation as I don't have any structural verilog netlist with matching liberty files at hand to create a complete EQY example.

Cronus-38, 2025-05-21:
For anyone else facing this issue in the future who is using the OpenRoad Flow Scripts flow, I was able to get my design to verify successfully using .lib cell descriptions (instead of verilog simulation models) by replacing the procedure write_eqy_script in ./scripts/load.tcl with the modified version below.  Note that the TC_LIB_FILES variable referenced in the procedure contains a list of .lib files that were used for synthesis, as defined in ./platforms/<LIBRARY>/config.mk 

proc write_eqy_script {} {
    set top_cell [current_design]
    set cell_files [get_verilog_cells_for_design]
    set outfile [open "$::env(OBJECTS_DIR)/4_eqy_test.eqy" w]
    # Gold netlist
    puts $outfile "\[gold]"
    puts $outfile "read_liberty -ignore_miss_func $::env(TC_LIB_FILES)"
    puts $outfile "read_verilog -sv $::env(RESULTS_DIR)/4_before_rsz.v \n"
    puts $outfile "prep -top $top_cell -flatten\nmemory_map\n\n"
    # Modified netlist
    puts $outfile "\[gate]"
    puts $outfile "read_liberty -ignore_miss_func $::env(TC_LIB_FILES)"
    puts $outfile "read_verilog -sv $::env(RESULTS_DIR)/4_after_rsz.v \n"
    puts $outfile "prep -top $top_cell -flatten\nmemory_map\n\n"

    # Recommendation from eqy team on how to speed up a design
    puts $outfile "\[match *]\ngate-nomatch _*_.*"

    # See issue OpenROAD#6545 "Equivalence check failure due to non-unique resizer nets"
    puts $outfile "gate-nomatch net*"

    # Necessary to avoid false positive after Yosys 0.49
    puts $outfile "gate-nomatch clone*\n\n"

    # Equivalence check recipe 1
    puts $outfile "\[strategy basic]\nuse sat\ndepth 10\n\n"
    # Equivalence check recipe 2
    puts $outfile "\[strategy sby]\nuse sby\ndepth 10\nengine smtbmc bitwuzla\n\n"

    close $outfile

}