YosysHQ/eqy #62: Make install not working

JIceberg, 2024-04-17:
Hello, I followed the instructions to install Yosys EQY on my Ubuntu 20.04 device as such:
```sh
$ git clone https://github.com/YosysHQ/eqy.git epy
$ cd eqy
$ sudo make install
```
Initially, it said that `yosys-config` had no such command, so I did some research and found that I had to install yosys-dev which I did as such:
```sh
$ sudo apt-get install yosys-dev
```
Then, when I run `sudo make install` I now get the following errors:
```
src/eqy_combine.cc: In member function ‘void {anonymous}::EqyCombinePass::print_ids(FILE*, Yosys::RTLIL::Module*)’:
src/eqy_combine.cc:82:25: error: ‘struct Yosys::RTLIL::Module’ has no member named ‘get_hdlname_attribute’; did you mean ‘get_bool_attribute’?
   82 |   for (string name : m->get_hdlname_attribute())
      |                         ^~~~~~~~~~~~~~~~~~~~~
      |                         get_bool_attribute
src/eqy_combine.cc:86:19: error: ‘ID’ has not been declared
   86 |    if (a.first == ID::hdlname)
      |                   ^~
src/eqy_combine.cc:96:16: error: ‘struct Yosys::RTLIL::IdString’ has no member named ‘isPublic’
   96 |    if (c->name.isPublic())
      |                ^~~~~~~~
src/eqy_combine.cc:99:27: error: ‘struct Yosys::RTLIL::Cell’ has no member named ‘get_hdlname_attribute’; did you mean ‘get_bool_attribute’?
   99 |     for (string name : c->get_hdlname_attribute())
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           get_bool_attribute
src/eqy_combine.cc:103:21: error: ‘ID’ has not been declared
  103 |      if (a.first == ID::hdlname)
      |                     ^~
src/eqy_combine.cc:114:16: error: ‘struct Yosys::RTLIL::IdString’ has no member named ‘isPublic’
  114 |    if (w->name.isPublic())
      |                ^~~~~~~~
src/eqy_combine.cc:117:27: error: ‘struct Yosys::RTLIL::Wire’ has no member named ‘get_hdlname_attribute’; did you mean ‘get_bool_attribute’?
  117 |     for (string name : w->get_hdlname_attribute())
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           get_bool_attribute
src/eqy_combine.cc:121:21: error: ‘ID’ has not been declared
  121 |      if (a.first == ID::hdlname)
      |                     ^~
make: *** [Makefile:26: src/eqy_combine.so] Error 1
```
Am I missing something or has something become outdated?

nakengelhardt, 2024-04-18:
Sorry, that section of the documentation is incomplete. EQY requires a matching Yosys version, as indicated by the [tags](https://github.com/YosysHQ/eqy/tags). We don't maintain the distribution packages so no clue what version you got there.

To use [strategy sby] you will also need the matching SBY version and its dependencies (solvers).

The [OSS CAD Suite builds](https://github.com/YosysHQ/oss-cad-suite-build/releases/latest) includes all the required dependencies (matching latest tool versions, solvers, etc), so I recommend using that instead of building from source unless you want to modify the source. (Having a compatible version of OSS CAD Suite at the end of your path is also a good way to fulfill the dependencies that you don't need to touch even when you are using your own build for EQY. There's an additional wrinkle that for the plugins to work the compile environment used needs to match the one yosys was compiled in, but we use an ubuntu docker so I think in your case you can locally compile and OSS CAD Suite yosys will be able to load the plugin. On other systems it might be required to use [this docker container](https://github.com/YosysHQ-GmbH/tabby-cad-plugin-build).)

JIceberg, 2024-04-18:
Ohh thank you so much! That makes a lot of sense.

closed 2024-04-18 by JIceberg