YosysHQ/eqy #65: EQY fails to prove a cell mapping equivalent

povik, 2024-04-26:
See the attached problem for EQY: [t2.tar.gz](https://github.com/YosysHQ/eqy/files/15134999/t2.tar.gz)

It is a purely combinational circuit mapped to standard cells. EQY is configured with

```
[gold]
read_verilog t2_gold.v

[gate]
read_liberty -ignore_miss_func t2_lib1.lib
read_verilog t2_mapping.v
hierarchy -top top

[strategy basic]
use sat
depth 10
```

and logs the following (redacted to the important parts)

```
EQY 20:11:10 [t2] Warning: Partition top.F contains 1 unused gold inputs.
EQY 20:11:10 [t2] Warning: Partition top.F contains 64 unused gate inputs.

...

EQY 20:11:10 [t2] run: Running strategy 'basic' on 'top.F'..
EQY 20:11:11 [t2] run: Could not prove equivalence of partition 'top.F' using strategy 'basic'

...

EQY 20:11:28 [t2] Warning: Failed to prove equivalence for 1/8 partitions:
EQY 20:11:28 [t2] Failed to prove equivalence of partition top.F
```

Meanwhile if I test with a Yosys script

```
read_verilog t2_gold.v
check -assert
design -stash gold

read_liberty -ignore_miss_func t2_lib1.lib
read_verilog t2_mapping.v
hierarchy -top top
flatten
check -assert

design -copy-from gold -as gold top
equiv_make gold top equiv
equiv_induct equiv
equiv_status -assert
```
the two netlists are proven equivalent.

The issue reproduces on the nightly OSS CAD Suite build: https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-04-26

jix, 2024-04-26:
Currently EQY requires the user to configure the partitioning in such a way that the proving the equivalence of each partition does not depend on invariants of the transitive fan in of the partition in question.

One simple way to ensure this (that doesn't scale to larger designs) is by including all of the gold side transitive fan in for each partition by adding this to the config:

```
[partition *]
amend *
```

This makes your example pass.

povik, 2024-04-26:
Ah so this is EQY functioning as intended. Thanks!

closed 2024-04-26 by povik

jix, 2024-04-26:
It's a known limitation, but not intended to stay this way.